b-bilko/Tracking-Turret #1: Tracking threads

DaveTheTroll, 2021-04-22:
Four threads continually running.
Communication between threads is three values.  X, Y and Fire.

1) Main thread.  Continually monitors video.  Looks for motion.  When it finds motion sets (X, Y) to centre of contour and Fire to True.  If it does not find motion it sets (X, Y) to zero and Fire to False.
2) X Thread.  Moves X motor towards X.
3) Y Thread.  Moves Y motor towards Y.
4) If Fire is true and current motor position is within tolerance of (X, Y), fires gun.

This will result in the ability to strafe and to be more responsive.

DaveTheTroll, 2021-04-23:
@b-bilko Threaded version pretty much done.

This is hopefully easier to read than the previous version as well as being more dynamic.  Capturing and analysis of images is completely asynchronous to movement and firing.

b-bilko, 2021-04-24:
I've had a look, and it mostly makes sense to me :D 

I'll do a test once I've got the hardware here, see if I can get things to move and click...

closed 2021-04-29 by DaveTheTroll