b-bilko/Tracking-Turret #5: Save Images when "triggered"

b-bilko, 2021-04-26:
Each time the trigger operates, save a few frames of video to a folder. 

Would need to have a cap on the number saved, or overwrite after a time to avoid filling the SD card up.

DaveTheTroll, 2021-05-29:
Currently images are logged.

Maximum rate 1 per second (so we don't fill up too quick).

Logged:
1) First image when tracking started.  This should be the empty static scene.
2) The first image when going into the motion state.  i.e. the first image when a target is found.  also after the target moves out of firing position.
3) whenever the gun is firing.

There is a maximum number of files allowed in the target folder.

DaveTheTroll, 2021-05-29:
@b-bilko