b-bilko/Tracking-Turret #19: Waveshare Stepper Driver

DaveTheTroll, 2021-05-06:


b-bilko, 2021-05-06:
Ran run_turret.py

Traceback (most recent call last):
  File "/home/pi/Tracking-Turret/run_turret.py", line 4, in <module>
    turret_with_config.go()
  File "/home/pi/Tracking-Turret/turret_with_config.py", line 41, in go
    turret = create()
  File "/home/pi/Tracking-Turret/turret_with_config.py", line 25, in create
    steppers = (stepper_waveshare.Stepper(X_DIR_PIN, X_STEP_PIN, X_ENABLE_PIN, X_MODE_PINS, MOTORS_REVERSED[0], "X"),
  File "/home/pi/Tracking-Turret/stepper_waveshare.py", line 9, in __init__
    self.motor.set_micro_step(DRV8825.STEP_ONE)
  File "/home/pi/Tracking-Turret/DRV8825.py", line 62, in set_micro_step
    self.digital_write(self.mode_pins, microstep[step])
KeyError: 0

b-bilko, 2021-05-06:
Repeat Testing: 

Code Ran, Did initial calibration of image. 

Error when movement identified. 

No Microswitch pin specified.  Skipping calibration.
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/Tracking-Turret/stepper_base.py", line 31, in __loop
    self.step(self.forward_dir if self.target - self.pos > 0 else -self.forward_dir)
  File "/home/pi/Tracking-Turret/stepper_waveshare.py", line 17, in step
    self.motor.set_direction(direction)
  File "/home/pi/Tracking-Turret/DRV8825.py", line 94, in set_direction
    self.digital_write(self.enable_pin, 0)
NameError: name 'self' is not defined
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/Tracking-Turret/stepper_base.py", line 31, in __loop
    self.step(self.forward_dir if self.target - self.pos > 0 else -self.forward_dir)
  File "/home/pi/Tracking-Turret/stepper_waveshare.py", line 17, in step
    self.motor.set_direction(direction)
  File "/home/pi/Tracking-Turret/DRV8825.py", line 94, in set_direction
    self.digital_write(self.enable_pin, 0)
NameError: name 'self' is not defined