b-bilko/Tracking-Turret #21: Threading in Adafruit hat

DaveTheTroll, 2021-05-17:
Multithreaded access to I2C for Adafruit results in sequential motion.

Use single thread for all Steppers on same Kit.

b-bilko, 2021-05-18:
============== RESTART: /home/pi/Tracking-Turret/run_turret.py ==============
No Microswitch pin specified.  Skipping calibration.
Traceback (most recent call last):
  File "/home/pi/Tracking-Turret/run_turret.py", line 4, in <module>
    turret_with_config.go()
  File "/home/pi/Tracking-Turret/turret_with_config.py", line 42, in go
    start(turret)
  File "/home/pi/Tracking-Turret/turret_with_config.py", line 38, in start
    turret.motion_detection()
  File "/home/pi/Tracking-Turret/turret.py", line 56, in motion_detection
    self.stepper_x.start_loop()
  File "/home/pi/Tracking-Turret/stepper_adafruit.py", line 25, in start_loop
    if not Stepper.threads.has_key(self.kit):
AttributeError: 'dict' object has no attribute 'has_key'

DaveTheTroll, 2021-05-18:
@b-bilko has_key error fix pushed.

b-bilko, 2021-05-18:
Seems to have pushed as I have a new Error: 

============== RESTART: /home/pi/Tracking-Turret/run_turret.py ==============
No Microswitch pin specified.  Skipping calibration.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/Tracking-Turret/stepper_adafruit.py", line 40, in __loop
    if s.__update():
AttributeError: 'Stepper' object has no attribute '_Stepper__update'

Traceback (most recent call last):
  File "/home/pi/Tracking-Turret/run_turret.py", line 4, in <module>
    turret_with_config.go()
  File "/home/pi/Tracking-Turret/turret_with_config.py", line 42, in go
    start(turret)
  File "/home/pi/Tracking-Turret/turret_with_config.py", line 38, in start
    turret.motion_detection()
  File "/home/pi/Tracking-Turret/turret.py", line 60, in motion_detection
    self.motion_sensor.find_motion()
  File "/home/pi/Tracking-Turret/motion_sensor.py", line 115, in find_motion
    base = self.get_empty_frame()
  File "/home/pi/Tracking-Turret/motion_sensor.py", line 74, in get_empty_frame
    frame, candidate = self.grab_image()
  File "/home/pi/Tracking-Turret/motion_sensor.py", line 54, in grab_image
    raise FrameGrabException()
motion_sensor.FrameGrabException

DaveTheTroll, 2021-05-18:
@b-bilko fix pushed

b-bilko, 2021-05-18:
============== RESTART: /home/pi/Tracking-Turret/run_turret.py ==============
No Microswitch pin specified.  Skipping calibration.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/Tracking-Turret/stepper_adafruit.py", line 40, in __loop
    if s.update():
TypeError: update() takes 0 positional arguments but 1 was given

Traceback (most recent call last):
  File "/home/pi/Tracking-Turret/run_turret.py", line 4, in <module>
    turret_with_config.go()
  File "/home/pi/Tracking-Turret/turret_with_config.py", line 42, in go
    start(turret)
  File "/home/pi/Tracking-Turret/turret_with_config.py", line 38, in start
    turret.motion_detection()
  File "/home/pi/Tracking-Turret/turret.py", line 60, in motion_detection
    self.motion_sensor.find_motion()
  File "/home/pi/Tracking-Turret/motion_sensor.py", line 115, in find_motion
    base = self.get_empty_frame()
  File "/home/pi/Tracking-Turret/motion_sensor.py", line 102, in get_empty_frame
    self.callback_nomotion(frame)
  File "/home/pi/Tracking-Turret/turret.py", line 73, in __on_no_motion
    self.stepper_x.set_target(0)
  File "/home/pi/Tracking-Turret/stepper_base.py", line 29, in set_target
    self.__get_flag.set()
AttributeError: 'function' object has no attribute 'set'
>>>

DaveTheTroll, 2021-05-18:
@b-bilko Fix pushed

b-bilko, 2021-05-18:
>>> 
============== RESTART: /home/pi/Tracking-Turret/run_turret.py ==============
No Microswitch pin specified.  Skipping calibration.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/Tracking-Turret/stepper_adafruit.py", line 40, in __loop
    if s.update():
TypeError: update() takes 0 positional arguments but 1 was given

Traceback (most recent call last):
  File "/home/pi/Tracking-Turret/run_turret.py", line 4, in <module>
    turret_with_config.go()
  File "/home/pi/Tracking-Turret/turret_with_config.py", line 42, in go
    start(turret)
  File "/home/pi/Tracking-Turret/turret_with_config.py", line 38, in start
    turret.motion_detection()
  File "/home/pi/Tracking-Turret/turret.py", line 60, in motion_detection
    self.motion_sensor.find_motion()
  File "/home/pi/Tracking-Turret/motion_sensor.py", line 115, in find_motion
    base = self.get_empty_frame()
  File "/home/pi/Tracking-Turret/motion_sensor.py", line 102, in get_empty_frame
    self.callback_nomotion(frame)
  File "/home/pi/Tracking-Turret/turret.py", line 73, in __on_no_motion
    self.stepper_x.set_target(0)
  File "/home/pi/Tracking-Turret/stepper_base.py", line 29, in set_target
    self.__get_flag().set()
  File "/home/pi/Tracking-Turret/stepper_base.py", line 22, in __get_flag
    return self.flag
AttributeError: 'Stepper' object has no attribute 'flag'
>>>

DaveTheTroll, 2021-05-18:
@b-bilko Fix pushed

b-bilko, 2021-05-18:
Steppers are now Energised but no movement

Video windows are loading as expected, tracking crosshair is appearing 

============== RESTART: /home/pi/Tracking-Turret/run_turret.py ==============
No Microswitch pin specified.  Skipping calibration.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/Tracking-Turret/stepper_adafruit.py", line 40, in __loop
    if s.update():
TypeError: update() takes 0 positional arguments but 1 was given

DaveTheTroll, 2021-05-18:
@b-bilko Pushed

b-bilko, 2021-05-18:
Tracking worked! Span really fast Anti clockwise... 

I've already set the range lower in the config file, currently set to: 

`MOTORS_REVERSED = (True, False)
MOTORS_RANGE = (50, 50)`

I didn't see an output in the debugging. post startup. 

See video..

b-bilko, 2021-05-18:
Also, I can't stop that without killing the Python IDE.

closed 2021-05-26 by DaveTheTroll